microbit-foundation/microbit-reference-design #3: Git Version control of this reference design is ugly

jaustin, 2017-01-05:
Coming from a software world where we get to see diffs of changes and have a series of commits for a large piece of work, using git for the HW project files feels painful and a little counter-intuitive.

This issue is a placeholder under which to collect comments and suggestions about how we can best manage and collaborate on a hardware design using Git/GitHub.

jaustin, 2017-01-05:
I'd be really interested to see if something like http://phdl.sourceforge.net/2.1/index.php could make the issue better - so the source of the design is stored here in git and we have scripts for generating things that can be used in other editors. This doesn't seem to handle layouts, though it might be a good start.

OwenBrotherwood, 2017-01-05:
https://easyeda.com/Doc/Open-File-Format/

OwenBrotherwood, 2017-01-06:
https://github.com/microbit-foundation/microbit-reference-design/issues/1#issuecomment-270845656 in context of using ascii and git and an acorn way.

OwenBrotherwood, 2017-01-07:
https://github.com/microbit-foundation/microbit-reference-design/issues/5#issuecomment-271101251