microbit-foundation/microbit-reference-design #4: contents of micro:bit reference design: web and repo

OwenBrotherwood, 2017-01-06:
From my analysis.

1. https://github.com/microbit-foundation/microbit-reference-design/tree/master/Altium
- should contain the master of the schematic design
- the ascii export, let us call it "ACCEL ASCII" should be included
- it is questionable if any Altium binary blobs for schematics give any meaning in the git repro
- it is questionable if Altium binary blobs for pcb give any meaning in the git repro
2. Concentrate on ensuring the schematic is correct
- remove all pcb designs until this is met and the web should only give a general idea of possibilities with perhaps the main goal of true shareability of schematic design to ensure pcb IRL.
3. Others like Eagle etc etc (the list is endless)
- document import for the interesting list of schematic/pcb sw
- ensure that the import works with active community support for checking
- "if it doesn't work, issue it" and concentrate on issues from people who are actually using the schematic/pcb sw instead of a broad range of possible sw users.

The above are just a few of my thoughts, dependant on feedback to 1 2 3 this issue may come into other areas.

sjwiseman, 2017-01-06:
1) I think that the Altium binary blob _should_ stay. It's the source, like it or not. Everything else is derived. 
Yes, any useful derived intermediate files should be published. I didn't think the  Accel was particularly useful. (It's also not something that  I can generate automatically from the outjob.)
2) Not at all convinced by this. The PCB is (imho) an important part of the design. If people want to generate derivative works rather than starting from scratch, it'll save a few days. It's also something to swipe footprints from). 
3) Yes! Import was surprisingly flaky and unsatisfactory. Given the maker movement's reliance on reuse, I'm surprised just how hard / bad it was. Obviously I'm spoilt with Altium's generally competent importers. 
I'm partly assuming that I'm doing it wrong(tm), and am hoping for assistance from seasoned users of those packages. 
I'm somewhat busy at the moment, but will  watch, and do what I can. Shout if there's anything in particular.

OwenBrotherwood, 2017-01-06:
1: Jep: let the blob be in: but never without a commit without the ASCII as documentation for change.
2: Jep: I am basically wanting more concentration on the sch to ensure 100% correct. I am not sure if anyone has actually used the pcb and tested: but I am sure someone will say something.
3: only with a little help from our friends, will the effort come forward: I do stuff as a total amatour and in many ways this gives a good result, relying on open mind and stubbornness.

At the present time: no immediate need for work by @sjwiseman : the documentation of how to export from Altium and import into a subset of interesting sw products is up to the community.

---

My current need for what I believe is "footprints", is the bottom connection layout on the micro:bit, otherwise I will have to do by hand :(

OwenBrotherwood, 2017-01-07:
- Good online:
 - https://circuits.io/
 - https://upverter.com
- Users use:
 - Eagle
- Fun use:
 - Fritzing
- Professional user:
 - Altium 

Export/import paths to be found

OwenBrotherwood, 2017-01-07:
Quick list of words

- collaboration
- break down of schematic to easy chunks ala. programing
- version control
- export from source
- community contribution to bring the ball forward
- 

and Play and Learn

OwenBrotherwood, 2017-01-08:
No need to join the channel as such, but possible read down from this link
https://microbit-community.slack.com/archives/hardware/p1483829409000042

OwenBrotherwood, 2017-01-08:
"Fork" and also ensure open format exports in pure ascii so that github is used for doc/progress/revisions.

[image]

OwenBrotherwood, 2017-01-08:
https://upverter.com/acornway/f52fb058b10f0566/acornway-reference-design/

OwenBrotherwood, 2017-01-08:
Export possibilities that are a available at the moment for upverter.com
- with due deference to Altium as source but possible way to another collaboration source with high attention to export to other tools 

https://upverter.com/acornway/f52fb058b10f0566/acornway-reference-design/#design-exports

https://forum.upverter.com/t/documentation-for-open-json-format/154

OwenBrotherwood, 2017-01-08:
[image]

sjwiseman, 2017-01-08:
Ah yes, I remember - rotated components all over the place!
Do you want to raise a ticket with Upverter?

OwenBrotherwood, 2017-01-08:
I think we wait a bit: I would prefer to break the diag. into smaller sections.
I am not sure of the terminology but there should be a way.
For programmers, reading all these lines of code (lines/pin nr etc) is hard: prefer to break into sections from start and reuse with no pain.

sjwiseman, 2017-01-08:
You're welcome to break the design apart, and stitch it all back together with hierarchy - but on a design this small and interconnected, I personally  like to have it all visible.  
(I suspect that hierarchy will also open a whole new can of worms when it comes to transferring between packages!)

OwenBrotherwood, 2017-01-09:
I concentrate on learning more import/export and the copper pattern of the edge connector.
Modular ref design could give "something", but as stated: could get wormy

OwenBrotherwood, 2017-01-09:
And seeing what footprints to swipe: but really missing the connector and edge copper
> The PCB is (imho) an important part of the design. If people want to generate derivative works rather than starting from scratch, it'll save a few days. It's also something to swipe footprints from).

sjwiseman, 2017-01-09:
On 9 January 2017 at 10:04, Owen Brotherwood <notifications@github.com>
wrote:

> And seeing what footprints to swipe: but really missing the connector and
edge copper

It's a Modulebit, not a Microtbit(tm). Designed to plug into breadboards,
designed to be a start point for your own project. If you want a  Microbit,
I'd strongly suggest you buy one!
You'll also note it's bigger, to give you some area on the PCB to build
stuff, without interfering with the basic functionality - just blow away
the audio section, and you have a decent amount of space.
No-fit the audio and the battery, and you've (just!) got enough space for a
little breadboard to stick on, then jumper wire to the 0.1" connections.
Or do something else, that's kinda the point :) I wasn't just cloning the
Microbit.

OwenBrotherwood, 2017-01-09:
Or a baggybit
https://github.com/microbit-drill/baggybit :)
Rapidly having a Pi 40 pin interface to pi accessories 
with SD flash and i2c extra io or capacitive touch and audio: if enough area
Keeping the present micro:bit v1 and hoping for more ram/flash in v2 and the edge copper on the back of the micro:bit having some i2c extras for i/o :)
[image]

OwenBrotherwood, 2017-01-09:
Found a nice verified Nordic nRF51822-QFAA with pcb
May not quite be a 	Nordic nRF51822-QFAA-R rev 3 as in tech.microbit.org but I hope it is good enough
https://upverter.com/acornway/9b6866459a4eb097/microbit-raspberry-pi-and-mbed-pins/

I search for other ref design components

OwenBrotherwood, 2017-01-10:
@sjwiseman @jaustin 
https://github.com/microbit-foundation/microbit-reference-design/issues/7#issuecomment-271550854 contains the first module of the mega:bit modular system.

OwenBrotherwood, 2017-01-11:
@sjwiseman I found out today that the ic on the ref diagram is not an ic for nRF51822 but a module, as you may have hinted at earlier in the thread (I didn't get the hint :) ).
https://upverter.com/acornway/150e8cc130b43e8d/megabit-Ratac-MDBT40-nRF51822-Module/ is done I think
https://upverter.com/acornway/65697ed30d7e35d3/megabit-nRF51822-QFAA-Module/ will document the implementation of the Ratac module

---

I have done some successfull imports of Eagle today: but Altium sch and pcb from this repro still does not work

OwenBrotherwood, 2017-01-11:
I will be testing plugins for Altium and eagle