microbit-foundation/microbit-reference-design #5: micro:bit reference schematic is upside down in Altium

OwenBrotherwood, 2017-01-06:
[image]

sjwiseman, 2017-01-06:
Can someone (anyone?) replicate this? Certainly doesn't do it here, either on the original or when I open the published version.

OwenBrotherwood, 2017-01-06:
@sjwiseman Can you make the original directly accessible to me by some means?
Also, which version of Altium are you using.
Then I do some work :)

sjwiseman, 2017-01-06:
It's just the red text that's upside down, by the way.  It's in the right place, just inverted. Odd!
It's good here in Altium 14, 16 and 17.freshest , and on 2 PCs across different sites. 

OK - some possible progress. Can you double-click some of the red text, and see if the 'Mirror' attribute is set?
It looks as if it's set by default, and ignored - on all versions I've got here. Maybe your eval licensed copy is actually respecting the attribute? Try turning it off, see what happens to the text. 
If this is it, I'll fix globally reissue. Very, very odd! (It looks like I set that attribute a few months ago and it's been applied ever since, with no visible effect. Sigh)
Have a version with the attributes cleared. 

[Modulebit_004a.zip](https://github.com/microbit-foundation/microbit-reference-design/files/690767/Modulebit_004a.zip)

OwenBrotherwood, 2017-01-07:
On the original ref design: reversing red text. Enclosed pic shows attrib.
I set to 0 degrees and no mirror and it looks good :)
I check with the 4a version 

[image]

OwenBrotherwood, 2017-01-07:
You are just gonna love this one: 180 changed to 0 gives ok text for 004a
[image]

OwenBrotherwood, 2017-01-07:
My thoughts at present time.
I would prefer the reference design to be a series of schematics that allow for a type of "object" approach that is know from programming.
The schematic is divided into well defined areas in separate schematics.
The schematics can then be combined into a pcb with the areas wished
Closer change control of sections of the design with ASCII and github and versioning

---

It may be "life" that the separate schematics have to be made in Eagle and checked with import into other programs
A clear community support for the work required is possible.
The pcb design being up to whoever has time.

worayoot, 2018-12-24:
From Altium 19
1. Select all (Ctrl+a)
2. Right click on string such as "Magnetometer, Accelerometer" -> Find similar object...
3. Orientation 180 Degree Select from 'Any' to 'Same' -> Click OK
4. At Properties window, change rotation to 0 Degrees and Justification select arrow up right
5. Done