microbit-foundation/microbit-reference-design #6: connectors / copper pattern / standard pcb components

OwenBrotherwood, 2017-01-08:
In general, the connectors of the micro:bit are a bit "hard"
The copper pattern, 4 holes and 1.25 stripes need to be a standard component to be cherry picked.
Connectors, 80 1.25 are quite often not available as standard components.

---

Some connectors I have found today: I will probably add to the list as time goes by
http://www.sullinscorp.com/catalogs/62_PAGE64-65_.125_HP_DS_WW_RA.pdf

jaustin, 2017-01-10:
For the connectors, that's more something for the dev-docs repo than the RD one, though I guess we might want to start collecting some useful micro:bit footprints here? Or perhaps in an additional repo called microbit-hw with things someone making accessories could use/need?

OwenBrotherwood, 2017-01-10:
Let us see what I find out of in this week for ensuring something structural and upkeepable

OwenBrotherwood, 2017-01-11:
In order to get parts, schematic and footprint for all micro:bit edge connectors, I have created a requests for https://upverter.com/concierge/ 
This allows for full micro:bit easy pcb creation if upverter put's in the work
I have asked for the following as I think they the ones needed
EBA40DCAN
EBA40DCBN
EBA40DCCN
EBA40DCWN
EBA40DCTN

I have to find a way to order the copper on the bottom of the micro:bit as a standard element for pcb

[image]

/cc @sjwiseman

OwenBrotherwood, 2017-04-12:
Proto has now eagle patterns for edge and connector @ https://github.com/proto-pic/micro-bit-eagle-libraries

OwenBrotherwood, 2017-04-12:
Anthony Kirby a KiCad schematic & footprint for the through-hole version of the edge connector socket
 https://github.com/anthonykirby/kicad_microbit_connector

OwenBrotherwood, 2017-04-12:
fritz parts for bottom connector and general pcb for bolt together
https://github.com/make-zurich/micro-bit-badge

[image]