microbit-foundation/microbit-reference-design #7: possible pin out problems or are there  different pin outs per package

OwenBrotherwood, 2017-01-09:
Nordic nRF51822-QFAA-R rev 3 (http://tech.microbit.org/hardware/) 
contra
https://upverter.com/eda/#tool=schematic,designId=65697ed30d7e35d3
- see datasheet pdf
example: Vdd 14 (ref design) is i/o in pdf

[image]

https://github.com/microbit-foundation/microbit-reference-design/blob/master/PDF/Schematic%20Print/Schematic%20Prints.PDF

OwenBrotherwood, 2017-01-10:
[image]

OwenBrotherwood, 2017-01-10:
The modules schematic is finished with relation to nordic doc.
I try and complete another module and find out how to connect sheets to one pcb
https://upverter.com/acornway/65697ed30d7e35d3/megabit-nRF51822-QFAA-Module/

rolandvs, 2017-01-10:
The VDD on pin 14 is wrong. The symbol is not correct.  The current reference is ok, Also check out/use the latest of the datasheets >=3.3...

jaustin, 2017-01-10:
The pinout used in the reference design is relating to the Raytac module, not the nRF51 SoC itself. So the pinouts are different, yes. Happy for me to close this issue given that explanation?

jaustin, 2017-01-10:
Closing this issue after creating #10 to track clarifying the problem

closed 2017-01-10 by jaustin

OwenBrotherwood, 2017-01-10:
The ref design uses modules:
http://tech.microbit.org/hardware/reference-design/

Module Choice
Because the micro:bit uses every pin on the nRF51822, only modules that exposed all of the GPIO of the chip could be used. We have started with the Raytac MDBT40, which is available from outlets like Seedstudio, and commonly used on things like Adafruit BLE boards.

```
This design is fully open source, and we’re happy to accept pull requests for variants that use other modules, or improvements.

For your own projects, using a different module is as simple as wiring the right pins. The hardware page has a detailed pinmap. Likewise, contributions to this pinmap document for other popular modules are welcome.

There is a list of nRF51882 modules maintained by Nordic Semiconductor, from which you could choose any module that has all 31GPIOs broken out. If your design doesn’t use all of the pins on the edge connector and you are able to recompile your software for your custom design (for example using mbed) then you could choose a range of other modules.

```
https://www.nordicsemi.com/eng/Products/3rd-Party-Bluetooth-low-energy-Modules