microbit-foundation/microbit-reference-design #9: MAG module, Freescale Semiconductor	MAG3110FCR1 in progress

OwenBrotherwood, 2017-01-10:
[image]

[image]

jaustin, 2017-01-10:
@OwenBrotherwood can you explain more clearly what the issue you're describing here is please? I can't see what you're suggesting I'm afraid... I presume this is to do with upverter import of the RD?

OwenBrotherwood, 2017-01-10:
yep: issue is general ensure that there is a base for all schematics/pcb for ref design.
The doc for creating is the two references, ref design plus another one.
I cherry pic from reference design

closed 2017-01-12 by OwenBrotherwood