microbit-foundation/microbit-reference-design #16: Waste o time

Bazmundi, 2024-10-24:
Tried openning the KiCAD versions, got blank PCB, schematic failed with missing libaries and a couple of other obtuse errors.  If you're not supporting, recommend you delete repository

martinwork, 2024-10-24:
Hi @Bazmundi, it seems to work for me.

I cloned this repo, launched KiCad, did File/Open Project and navigated to where the Kicad folder was cloned to find the .pro files. 

The total of my knowledge of KiCad is that the installer took a really long time! 

Is there something else needed to make it go wrong?

[image]

[image]

martinwork, 2024-10-24:
Oh! I can see the PCB looks blank. On trying to update from schematic... there are dialogues that I don't know how to answer and a list of missing symbol libraries.

These files were produced 8 years ago, presumably from the current version of KiCad. Maybe KiCad has moved on such that it has a problem loading old files? Certainly I don't know how to use it! Perhaps try an older version?

I tried Kicad v4.0.7, in which I could see the schematic in Eeschema. When I tried to see the pcb, there was a message about installing libraries (should have taken a picture - it only shows once!). 

I feel it's likely these difficulties are caused by the KiCad UI, not by the files saved here. I need to read the getting started guide...